mihaigalos/miniboot #3: Error in the generated bootloader file made by thet Makefile

obitola, 2018-03-21:
pi@raspberrypi:~/XX-Core/arduino/miniboot/src $ make rebuild

-------- start --------

Cleaning project:
rm -f miniboot.hex
rm -f miniboot.eep
rm -f miniboot.obj
rm -f miniboot.cof
rm -f miniboot.elf
rm -f miniboot.map
rm -f miniboot.obj
rm -f miniboot.a90
rm -f miniboot.sym
rm -f miniboot.lnk
rm -f miniboot.lss
rm -f miniboot.o I2C-master-lib/i2c_master.o  
rm -f  miniboot.lst I2C-master-lib/i2c_master.lst
rm -f miniboot.s I2C-master-lib/i2c_master.s
rm -f miniboot.d I2C-master-lib/i2c_master.d
rm -f .dep/* 
rm -f bootloader.h
Errors: none
-------- end --------

avr-g++ (GCC) 4.8.1
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: miniboot.c
avr-g++ -c -mmcu=atmega328p -I. -g   -Os -pedantic -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Wall -Wstrict-prototypes -Werror -Wa,-adhlns=miniboot.lst  -std=gnu++11 -DF_CPU=8000000UL -MD -MP -MF .dep/miniboot.o.d miniboot.c -o miniboot.o 
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
In file included from miniboot.c:6:0:
bootloader.h:2:1: error: expected unqualified-id before ‘-’ token
 -en 
 ^
Makefile:258: recipe for target 'miniboot.o' failed
make: *** [miniboot.o] Error 1

mihaigalos, 2018-03-22:
can you please list the contents of bootloader.h?

obitola, 2018-03-28:
Sorry for the delayed response. 
These are the contents of the bootloader.h file

  1 // Automatically generated by Makefile. DO NOT EDIT!
  2 -en
  3 
  4 #pragma once
  5 -en
  6 
  7 #define BOOTLOADER_START_ADDRESS 0x7800
  8 #define EEPROM_CONFIGURATION_START_BYTE 0x03F6
~

mihaigalos, 2018-03-29:
it looks like the bootloader.h was not correctly generated by make's running of the Makefile. Did you edit the Makefile?

mihaigalos, 2018-06-03:
Which system are you on? Linux? Are you using make to regenerate the bootloader.h? Does the file get regenerated?

lukewaski, 2018-06-03:
> Do: "mihaigalos/miniboot" <miniboot@noreply.github.com>
> DW: "lukewaski" <admin@3dr.org>; "Comment" <comment@noreply.github.com>
> Data: 2018-06-03 19:15:59
> made by thet Makefile (#3)

Yes - Im on linux. Tried on raspbian and Debian.
I'm using make rebuild, the bootloader.h file is beeing gerated.

------ Wiadomość oryginalna ------

luckwaski, 2018-06-13:
Thank you - I can confirm it compiles fine under Linux now.

closed 2018-06-14 by mihaigalos