mihaigalos/miniboot #4: Bootloader corrupting code

luckwaski, 2018-07-06:
Hello,
for few days now I'm debuging a weird issue with miniboot. I'm attaching arduino sketch file + hex sources that will fill i2c memory with a program. Then after a bootloader is loaded to device it starts loading data to internal flash, and then the soft wont start (it just loops on main() of the miniboot as theres nothing to update due to timestamps).

By downloading flash to hex I have noticed that 2 bytes of the code sitting in flash are beeing changed. Whats funny - its always the same bytes, and the resulting "corruption" always have the same values..

What bootloader put in flash:
`:200000000C94**003C**0C9452010C9452010C9452010C9452010C9452010C9452010C9452015F`

What does the original data in eeprom look like:
`0C94**2A01**0C9452010C9452010C9452010C9452010C9452010C9452010C945201`

Bolded the issue.
The atmega, and eeprom are fine.
Any idea?

Thank you
[what_bootloader_put_in_flash.zip](https://github.com/mihaigalos/miniboot/files/2170501/what_bootloader_put_in_flash.zip)

[test.zip](https://github.com/mihaigalos/miniboot/files/2170496/test.zip)

mihaigalos, 2018-07-06:
Hi luckwaski,

miniboot is doing exactly as it should. If you check out the datasheet [1]
[2], the address 0 (reset) is actually a jump to the application code. In
this case, a jump to miniboot!

It does this by taking in the original application and modifying only the
first 2 bytes, the ones you identified! If it wouldn't, miniboot would
never run again. The 003C is the machine code to jump to the default
address of miniboot, futher up in the Flash region.

See, you've already learned something useful today!

Mihai

[1]
http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42735-8-bit-AVR-Microcontroller-ATmega328-328P_Datasheet.pdf#page=72&zoom=auto,-214,743
[2]
http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42735-8-bit-AVR-Microcontroller-ATmega328-328P_Datasheet.pdf#page=72&zoom=auto,-214,743

closed 2018-07-09 by mihaigalos

Fabiancj81, 2019-03-07:
Hi friend
  thanks for your answer, since I have the same problem, but now the next question comes up, why does not the main program run me? I have reviewed the flash memory map and it is correct (it only has the byte mentioned above as the only difference). my main program has a size of 22Kb.
best regards
Fabian