mihaigalos/miniboot #16: TRAVIS_PULL_REQUEST

chaifeng, 2019-10-24:
https://github.com/mihaigalos/miniboot/blob/995c22f4e18acbf5417e3205a0f455f0134b3107/scripts/post_summary.sh#L14

Should check the value of `TRAVIS_PULL_REQUEST`, it may be `false`.

[Environment Variables - Travis CI](https://docs.travis-ci.com/user/environment-variables/#convenience-variables)
> TRAVIS_PULL_REQUEST is set to the pull request number if the current job is a pull request build, or `false` if it’s not.

chaifeng, 2019-10-24:
And also, we can put CR in a string directly

for example, change the following code

    header="\`CI Auto Message\`\n---\n"

to

    header="CI Auto Message

    "

We cannot use `\n` everywhere, it depends on who uses it.

mihaigalos, 2019-10-24:
> Should check the value of TRAVIS_PULL_REQUEST, it may be false.

Good catch @chaifeng ! I'll create a fix for this.

> And also, we can put CR in a string directly

This contains the "  ` " character which is used by markdown. I like the one-line solution better! :blush:

chaifeng, 2019-10-24:
One-line solution:

    header='`CI Auto Message`'$'\n'$'\n'"---"$'\n'

Use `$'\n'` to replace c-style CR `'\n'`, don't put `$'\n'` in quotes.

mihaigalos, 2019-10-24:
Looks nice, I find the initial syntax more intuitive, even with escaped back-quotes. :laughing:

closed 2019-10-24 by mihaigalos

chaifeng, 2019-10-24:
If you like the C-style syntax

    printf -v header "\`CI Auto Message\`\n---\n"

Using `printf` to save a C-style string to the variable `header`, equals to:

    header="\`CI Auto Message\`
    ---
    "

mihaigalos, 2019-10-24:
But what's the difference between these two following? I mean they're the same, the first is just extra calling `print -v` ?

```bash
printf -v header "\`CI Auto Message\`\n---\n"`
```
to the initial:
```bash
header="\`CI Auto Message\`\n---\n"`
```

chaifeng, 2019-10-24:
Yes, they are similar, what about this?

    printf -v push_message '`%s`\n---\n%s\n%s' "$header" "$body" "$footer"

mihaigalos, 2019-10-24:
Pure. Implemented in #19.

PR #17 (2019-10-24): Fix post summary when no PR was created
Fixes #16, which occurs when a change triggers a CI job but no PR was created. So there is no endpoint for `curl` to post the summary to.
PR #18 (2019-10-24): Fix inconsistency in variable type which is bool
Fixes #16, handles case where variable is `false`.