mihaigalos/miniboot #23: LED just blinks quickly, program never flashes

nullstalgia, 2020-01-01:
Hi! I think I'm doing something wrong here and could really use some help

I'm wanting to use this on an ATMega32u4 (Arduino Micro) later on, but I wanted to make sure I can get everything working on a 328P first. I'm using a Nano for these tests.

After uploading the given .hex for the UART method, I used CuteCom (Linux) to upload a basic Blink sketch, and using E2PROM::dump, I get this
[image]

After burning the compiled miniboot.hex with the given fuses in miniboot.c, the led on PB0 (Arduino pin 8) just blinks forever, around 4x a second?

I'm pretty stumped and could use any advice you can muster.

nullstalgia, 2020-01-01:
I took a page out of another issue's author and I made the LED_TOGGLE for CRC and Flashing seperate, and it seems that it's just doing the CRC forever?

nullstalgia, 2020-01-01:
Hey, it turns out that the YAT Terminal on Windows actually works in conjunction with the awesome python script in the repo. I'm probably gonna build something for the ESP32/8266 to make it a bit easier/OS agnostic :)

mihaigalos, 2020-01-01:
Hi @nullstalgia, so I cannot really tell you what is happening. I suspect a fuse problem.

You could try:
* using Fuses :     Ext: 0xFF, Hi: 0xD8, Lo: 0xE2
* after you flash the external i2c eeprom using the uart method, write it to serial and compare the is-should be hex files.
* dump the contents of the *internal* eeprom and see if it contains a linux timestamp. If it matches the timestamp of the firmware in the i2c eeprom, miniboot will skip it (avoids unnecessary writes to flash).

As I've said, I am really suspecting a fuse problem here. If you can get it up and running, I need to update the readme with these fuse settings.

Yes, the python script is awesome. But if you have any further improvements, in particular for linux, please fork and open up a PR. I highly value any contributions!

nullstalgia, 2020-01-04:
https://github.com/nullstalgia/esp32_eeprom_flasher

Been working on this hard for the last few days, and now I think it's *okay* enough to be shown off/tested.

Let me know if you like it/want to link to it in your repo. :)

mihaigalos, 2020-01-05:
> Let me know if you like it/want to link to it in your repo. :)

Awesome. I've ref'ed it via submodules.

What about the original problem and the ticket? Did you manage to get it up and running correctly?

nullstalgia, 2020-01-05:
Yep, it turns out that cutecom doesn't properly send files, and miniboot just keeps trying the crc loop/doesn't leave it. But with the python script/this tool, it uploaded successfully and was able to be read by Miniboot. :)

closed 2020-01-05 by nullstalgia