mihaigalos/miniboot #25: Switch I2C address with switch

nullstalgia, 2020-01-08:
Hey, me again :)

So originally I was using an AT24C256, but I grabbed some CAT24M01's, and noticed that they have 2 "partitions" it seems.

I removed the `constexpr` qualifier from the address definition, added the i2c_address to `isReflashNecessary`, and... the code doesn't fit in the bootloader section.

I could double the word count with fuses, but I'm wondering if I can scrape every byte I can with some neat tricks. 

In io.h I added

```
#define SWITCH_INPORT(name) CONCAT(PIN, name)
#define SWITCH_OUTPORT(name) CONCAT(PORT, name)
#define SWITCH_INIT() OUTPORT(SWITCH_PORT_NAME) = OUTPORT(SWITCH_PORT_NAME) & ~(1 << SWITCH_PIN); SWITCH_OUTPORT(SWITCH_PORT_NAME) &= ~(1 << SWITCH_OUTPORT)
#define SWITCH_READ() SWITCH_INPORT(SWITCH_PORT_NAME) & (1 << (SWITCH_PIN))
```

And changed `source_i2c_address_for_program` depending on SWITCH_READ

If you can give any input, I'd be grateful!

mihaigalos, 2020-01-08:
> Regardless of the chips used, it would also be nice to have an easy way to switch the address via a switch instead of reuploading the bootloader.

No ideea what you mean here.

> I removed the constexpr qualifier from the address definition, added the i2c_address to isReflashNecessary, and... the code doesn't fit in the bootloader section.

Code diff or it didn't happen.

> but sometimes when erasing flash, it would wipe the bootloader

You can use the lock bits to prevent wiping the bootloader section.

> but I'm wondering if I can scrape every byte I can with some neat tricks.

Code or discussion please. :)

> And changed source_i2c_address_for_program depending on SWITCH_READ

No ideea what you mean here, sorry. What exactly are you trying to accomplish? Please give a better explanation.

nullstalgia, 2020-01-09:
Apologies if I wasn't clear. My goal is to have the AVR read a pin to see if it is being driven HIGH or LOW, and to change the I2C address based on that.

https://github.com/nullstalgia/miniboot/commit/7ec60abefd09c97c3511bcf1ce00fc800ef96765
This commit includes those changes plus having 2 separate LEDs for reading checksum/writing to flash

But I also ran a test with just the one LED, and had the same result, which was the resulting code tried to read instructions from addresses that go past 0x8000 (32kb, the maximum flash size).

```
avrdude: reading input file "miniboot.hex"
avrdude: input file miniboot.hex auto detected as Intel Hex
avrdude: ERROR: address 0x8010 out of range at line 129 of miniboot.hex
avrdude: read from file 'miniboot.hex' failed
```

I was wondering if you had any insight on finding a more efficient way to change the address during runtime to fit in the 1024 word limit. :)

mihaigalos, 2020-01-11:
OK I see.

> I removed the constexpr qualifier from the address definition, added the i2c_address to isReflashNecessary, and... the code doesn't fit in the bootloader section.

That's because the address is baked into the opcode when compiling with `constexpr` versus reading it from flash without it (your implementation, which is more inefficient but versatile for accounting for the partitions).

I have optimized miniboot a lot and do not know of any way of reducing the resulting ~2KB binary even further. Here are your options:
* compile with debugging symbols, look at the program map and see how much every function needs. Find out which ones take up most space and further refine them, possibly by refactoring them to assembly. Have a look at my [SoftwareUart](https://github.com/mihaigalos/Drivers/tree/9d97384c3b61ded51e7cc332b0ca9c9f5db88454/AVR) implementation for info on how to mix assembly and C.
* increase the bootloader size to 2048 words by tweaking the Lo and Hi fuses. This would leave less space for the application then.

Hope this helps, let's discuss how to proceed further if you get any further ideeas!

mihaigalos, 2020-01-11:
.. but really I need to be asking _why_ you need to read from separate partitions in the external I2C eeprom? I mean you can have multiple firmare versions all served by the 0x51 I2C slave address and different start offsets in them. The start offset is already in the implementation of miniboot, so you just need to update that field depending on which firmware you want miniboot to fetch from the external eeprom.

nullstalgia, 2020-01-13:
Sorry, I missed your replies!

So you're wondering why I would bother switching the I2C address when I can switch the `eeprom_address_offset` during runtime instead?

That *could* work, but some other (cheaper) EEPROMs only have 32k of space, and to make sure that the bootloader can be universal/handle programs larger than 16k with any EEPROM, having it switch the actual address could save future frustration. 

However, I did do some testing with your idea and it does seem to work just fine as well.

Another thing I have been considering is having 2 cheaper EEPROMs with an Address pin connected to a Switch so there will always be a slave on address 0x50 and 0x51, but they change places based on the switch. (Although, I'm not too sure how it would work with more than 2 EEPROMs...)

---

Unfortunately, I personally don't have the skills to refactor using assembly.

Also it turns out the fuses I have set have 2048 words allocated, so I do have room for these changes and my normal code.

--- 

In summary:

Both ideas seem to work, even when messing with the Switch during startup .

(But I'm not too sure I know *why* it resolves itself. It seems to restart the bootloader after `writeFlashFromI2C` finishes running. I haven't looked deep into it however)

I do have room for either change with 2048 words (4096 bytes).

Additionally, hardware solutions can be found using the Address pins of the EEPROMs, avoiding this issue altogether.

mihaigalos, 2020-01-13:
> Another thing I have been considering is having 2 cheaper EEPROMs with an Address pin connected to a Switch so there will always be a slave on address 0x50 and 0x51, but they change places based on the switch.

Most I2C devices have the possibility to change the address, using address pins as you said. You don't need a switch for this. Depending on how you connect the address pins (VDD or GND), you change _some_ of the address bits to 0 or 1.

> Additionally, hardware solutions can be found using the Address pins of the EEPROMs, avoiding this issue altogether.

Yes, precisely. Please bear in mind that this solution still require a variable for the address which cannot be constexpr anymore, hence you need the 2048 word space for miniboot. 

------

As far as I can see CAT24M01 contains 1Mb (128kB) of eeprom. Miniboot uses uint16_t in the source [implementation](https://github.com/mihaigalos/miniboot/blob/2f089bc98748a40530d4a14a5d6d378fade3385b/src/i2c_communication.h#L6) for reading from the external eeprom. Which means, as it stands, it can address up to 64KB. You can easily fit 2 full applications at least in this bad boy and use miniboot to flash whichever of them you need.

Let's continue the discussion if there are still unknowns. I will close this issue for now.

closed 2020-01-13 by mihaigalos